Snapmaker/SnapmakerCuraPlugin #25: Custom Materials do not use correct print temperature

evilC, 2022-12-12:
When adding a custom material for the J1, it does not use the correct printing temperature.

Repro:

1. A FRESH WINDOWS INSTALL was used (22H2, Build 19045.2251) just to be sure nothing else was causing the problem
2. Install Cura 5.2.1
3. Add J1 plugin
4. Start Cura
5. Add a Generic FFF printer with 2 extruders
6. Add a J1 printer and set it as the current printer
7. (Screen recording begins at this point)
8. Open `Material Preferences` (Ctrl-K)
9. Click `Create New`
10. Add `1` to the end of custom material name
11. Go to `Print Settings` tab
12. Set `Default Printing Temperature` to `234`
13. Close `Material Preferences` - new material is currently selected
14. ERROR: `Printing Temperature` reads `215` (Should be `234`)
15. Set `Generic FFF Printer` as current printer
16. Open `Material Preferences` (Ctrl-K)
17. Click `Create New`
18. Add `2` to the end of custom material name
19. Go to `Print Settings` tab
20. Set `Default Printing Temperature` to `235`
21. Close `Material Preferences` - new material is currently selected
22. CORRECT: `Printing Temperature` reads `235`

[image]

As is obvious from the screen recording, no such problem exists with a Generic FFF printer profile, so I can only assume that the error is because of the Snapmaker plugin

Repro'd on:

- Two computers (Fresh install and my normal PC)
- Cura 5.2.1, 5.0.0 and 4.5.1
- Plugin version 0.3.0 and 0.2.1

parachvte, 2022-12-12:
It's probably because we set a global profile with 215°C default temperature.

Thanks for the feedback, I'll check it later.

parachvte, 2022-12-13:
@evilC

I publish a pre-release version [0.4.0](https://github.com/Snapmaker/SnapmakerJ1CuraPlugin/releases/tag/0.4.0).
Could you check again if custom material's parameters applied correctly?

closed 2022-12-13 by parachvte

evilC, 2022-12-13:
> @evilC
> 
> I publish a pre-release version [0.4.0](https://github.com/Snapmaker/SnapmakerJ1CuraPlugin/releases/tag/0.4.0). Could you check again if custom material's parameters applied correctly?

Confirm fixed in 0.4.0

PR #27 (2022-12-13): Improvement: Separate material parameters to material definition
Closes #25