Snapmaker/SnapmakerCuraPlugin #28: J1 profile does not import correct values for custom materials

evilC, 2022-12-13:
Plugin version: 0.4.0

Whilst #25 is now fixed, there are still issues.
If you have existing custom materials, when you add a J1 printer, it does not have the correct temperature for the custom materials.

Repro:

1. FRESH install of Cura 5.2.1 (With old 5.2.1 settings folder deleted)
2. Install Cura 5.2.1 and run
It is worth noting that other versions of Cura are still installed, so upon install, existing custom materials and printer profiles are imported
3. Install J1 plugin 0.3.0 and restart Cura
4. (Screen recording begins here)
5. Observe that Ender 3 profile has been imported, and existing custom materials (TPU 98A at 240deg printing temp is used in this example)
6. Add a J1 printer
7. BUG: TPU 98A material has a printing temp of `200`
8. Switch to Ender 3 profile
9. TPU 98A material is still at `240` temperature for the Ender 3

It is worth noting, however, that if I add a different printer (eg a Creatiliy CR-10), then I see the same thing. So I am not 100% sure that this is a bug with the J1 plugin, it's maybe more a quirk of Cura?

parachvte, 2022-12-14:
There is no screen recording uploaded here. 🤔 

If you are using Ender 3 printer, then it's totally Enter 3's profile that are taking effect (maybe it's something wrong with ender 3 / CR10?).

We will check it later this week.

evilC, 2022-12-14:
Apologies, I added the missing screen recording that I recorded yesterday.
When I mentioned the Ender 3, I was saying that the issue does **not** happen with the Ender 3 profile, but **does** happen with the J1 profile.
However, I just tried to repro this again today and failed :(

parachvte, 2022-12-19:
After several failures to re-produce this, I found the clue.

Checkout [this file](https://github.com/Ultimaker/fdm_materials/blob/master/generic_pla_175.xml.fdm_material#L87).
It's the definition of `Generic PLA 1.75mm`, in L87, you can see there are custom settings for Creality3D machines.

If you modify the settings of the material in UI (e.g. set `Default Print Temperature` to `210`), it only modify the custom settings for your Ender 3. The default settings for the material remains `200`. Other printers using this material will still using the its default settings (e.g. `200`).

What you want to do is to modify the print temperature for all printers (but not only for ender3), right?

Here is how to do it right: always use clean material which doesn't have machine specific custom settings.

In your case, never duplicate material from `generic PLA`, `generic ABS` and `generic PETG`: https://github.com/Ultimaker/fdm_materials/search?q=creality_ender3

parachvte, 2022-12-19:
I'm closing this issue. Feel free to re-open it if you find other cases.

closed 2022-12-19 by parachvte

evilC, 2022-12-19:
Oh, well done, sir!
Apologies for wasting your time on this.
These quirks of Cura are really annoying - there's a bunch of stuff like this in Cura where it's not made obvious to the user what is actually going on under the hood.