Snapmaker/SnapmakerCuraPlugin #29: Right extruder does not start heating until Left extruder and bed are up to temperature

evilC, 2022-12-13:
Plugin version: 0.3.0

The `M104 T1` command is after the `M190 S` command, so the right extruder does not start heating up until the bed is at full temperature.

In this example, left extruder (T0) is set to `200`, right extruder (T1) is set to `160` and the bed is set to `60`

```
								;Generated with Cura_SteamEngine 5.2.1
								T0
								M82 ;absolute extrusion mode
								;--- Start G-code Begin ---
> Heat T0							M104 S200 ;Set Hotend Temperature
> Heat bed							M140 S60 ;Set Bed Temperature
								G28 ;Home
								G1 Z0.8
> Wait for T0 to heat up					M109 S200
> Wait for bed to heat up					M190 S60
								G1 Z0.8 F6000
								M201 X10000 Y10000 Z500 E5000
								M593 P1 F50
								M205 V5
								G92 E0
								G1 F200 E2
								G92 E0
								;--- Start G-code End ---

								G92 E0
								M2000 S200 V250 A6000
								G92 E0
								G1 F1500 E-6.5
								;LAYER_COUNT:784
								;LAYER:0
								M107
> Heat T1							M104 T1 S160
								M204 S1000
								G1 F600 Z1.28
								G0 F6000 X134.82 Y65.098 Z1.28
								;TYPE:SKIRT
								G1 F600 Z0.28
```

evilC, 2022-12-13:
If you modify the start G-code and add `M104 S{material_standby_temperature, 1} T1 ; Start heating T1` after the line `M140 S{material_bed_temperature_layer_0} ;Set Bed Temperature`, then this somewhat fixes the issue.
HOWEVER, it has an undesired side-effect:
If you do a single-extruder print, the 2nd extruder is still heated to it's standby temperature, even though it is not used.
I have [asked on the Cura forums](https://community.ultimaker.com/topic/42287-start-g-code-to-prehead-2nd-extruder/) if there is a workaround for this

parachvte, 2022-12-14:
This is regarded as a feature/bug of the slicer. It only starts to heat up T1 until first time switch.

I think it's not elegant at all to workaround this by start code.

evilC, 2022-12-15:
Well you already have it for T0. I have not tried it, but I would imagine if you sliced in Cura with the left extruder disabled (ie a print that just uses the right extruder), then as it stands, I think the printer would heat up the left extruder (Even though it is not used in the print) and wait for it to hit print temperature, before starting to heat the right extruder (Which **is** used in the print) because that is what the `Start GCode` in the plugin tells it to do.

parachvte, 2022-12-15:
Nah.

The `Start GCode` tell it to use `M104` or `M109` to heat up. The slicer will decide which extruder to use, and prepend `T0` or `T1` command based on your first layer setting.

evilC, 2022-12-15:
Ah OK, thanks

closed 2023-04-12 by parachvte