Snapmaker/SnapmakerCuraPlugin #31: Confusing reporting of right extruder material and temperature in Copy Mode

evilC, 2022-12-15:
1. Slice in Cura with Snapmaker PLA on left extruder, Snapmaker ASA on right extruder. Right extruder is **disabled** in Cura for this print.
2. Send file to printer
3. Select file on printer's touchscreen
4. Select Copy Mode for print
[image]
Notice under `Nozzle Temp`: `L:215 R:255`  
This would be correct for `Normal` mode, but for `Copy` mode, it always uses the same temp
5. So for `Copy` mode, it should show the same filament and temp for both L and R extruder, to avoid confusion
6. Start print
[image]
Everything is correct in this screen, no problem

lademeister, 2023-01-12:
Do you mean that in copy mode it should ignore the temp setting for ASA in your example and instead use the same temperature as for the left extruder?
I don’t agree then, because that would only be useful if you actually changed to the same filament on the right extruder.

In case the same bed temperature works for two different filaments, why wouldn’t you allow to print in copy mode using different filaments?

Of course you have a gcode then that would always require those filaments if you reuse it later unless you manually change the extruder temp when using copy mode.
But that’s pretty much the same when you re-use a gcode that was generated for another filament anyhow.

parachvte, 2023-01-12:
The best solution would be making temperatures for both extruders configurable.

Considering G-code already contains M104 / M109 when switching extruders, it would need parse of G-code and replace them to expected temperatures.

For different filament, bed temperature & retraction parameters are required too, and it's difficult to change them during print, and the printer needs to support independent extrusion.

evilC, 2023-01-12:
@lademeister 
> Do you mean that in copy mode it should ignore the temp setting for ASA in your example and instead use the same temperature as for the left extruder?
I don’t agree then, because that would only be useful if you actually changed to the same filament on the right extruder.

On the first screen, it says that it is going to heat the left extruder to 215 and the right extruder to 255  
When you actually start the job, the right extruder is headed to 215, not 255

> In case the same bed temperature works for two different filaments, why wouldn’t you allow to print in copy mode using different filaments?  

Yes, this would be an acceptable solution also, seems like a perfectly valid use-case to me. However this is not what actually happens, and could of course lead to a jam in the right extruder because the filament is not heated up enough.
However, in this case, it might be a little confusing in the slicer, because for the vast majority of prints, users are likely to be using the same filament for both extruders. In my example above, I sliced in Cura, and the right extruder was disabled - so I would reasonably expect that when I sent the print to the printer and enabled mirror mode, it would use the same temperature for both extruders. If your suggestion were implemented, there would need to be an extra step on the printer to allow the user to select the temperature for the 2nd extruder that was baked into the GCode

However, it is worth noting (@parachvte ) that in this case, there is no `M104` command in the GCode that sets the right extruder to 255deg. The printer is actually pulling that information from the commented header section at the start of the GCode.  
To clarify, the information on the screen is being pulled from the section at the start which looks like this:
```
;Header Start
;Version:1
;Slicer:CuraEngine
;Printer:Snapmaker J1
;Estimated Print Time:4524
;Lines:125771
;Extruder Mode:IDEX Full Control
;Extruder 0 Nozzle Size:0.4
;Extruder 0 Material:Snapmaker PLA White
;Extruder 0 Print Temperature:215
;Extruder 1 Nozzle Size:0.4
;Extruder 1 Material:Snapmaker ABS Black
;Extruder 1 Print Temperature:255
;Bed Temperature:65
```
ie there is no `M104 T1 255` command in the GCode, it is instead using the line `;Extruder 1 Print Temperature:255` to populate the temperature for extruder 1 on the screen of the printer

parachvte, 2023-01-12:
> However, it is worth noting (@parachvte ) that in this case, there is no `M104` command in the GCode that sets the right extruder to 255deg. The printer is actually pulling that information from the commented header section at the start of the GCode.

Yeah you are right. I updated my second paragraph.

We all agree with the original idea, displaying both left & right extruder (e.g. 215°C) to the same temperature for copy/mirror mode. Users need this information. I was extended the idea to configure both extruders, there are things to consider and that makes it a bit complicated.

evilC, 2023-01-17:
> Yeah you are right. I updated my second paragraph.
>
> We all agree with the original idea, displaying both left & right extruder (e.g. 215°C) to the same temperature for copy/mirror mode. Users need [...]

I agree. I think you guys probably have bigger things on your plate for now than catering for that edge case. I think just correcting this issue (I guess just update the plugin to omit the information in the header for the 2nd extruder if it is not used in the print?)