Snapmaker/SnapmakerCuraPlugin #32: Snapmaker J1 plugin for Cura not connecting to J1 printer over wifi

alziepen, 2022-12-19:
After downloading the latest plugin 0.4.0 release, and Cura version 5.2.1, Cura is unable to connect to printer and gives a "could not connect to device" description after trying to connect via entering the IP address in manually.

Operating on windows 10 pro, version 10.0.109044 Build 19044. The hardware is a dell precision M6800 workstation laptop
attempting to troubleshoot as it's apparently connected aromatically according to other users.
I've tried turning the wifi of the machine off and back on again. The firmware of the J1 is up to date. 
I've also tried uninstalling Cura, and downloading both Cura and the plugin on the same day which also didn't work.

[FW_0.log](https://github.com/Snapmaker/SnapmakerJ1CuraPlugin/files/10256429/FW_0.log)
[SC_0.log](https://github.com/Snapmaker/SnapmakerJ1CuraPlugin/files/10256430/SC_0.log)
[SC_1.log](https://github.com/Snapmaker/SnapmakerJ1CuraPlugin/files/10256431/SC_1.log)
[SC_2.log](https://github.com/Snapmaker/SnapmakerJ1CuraPlugin/files/10256432/SC_2.log)

parachvte, 2022-12-19:
You need add a non-networked Printer -> Snapmaker -> Snapmaker J1.

[image]

alziepen, 2022-12-19:
Ok thank you for clarifying. I should have checked to see if the printer popped up in little export popup window. For whatever reason within the facebook group something got lost in the discussion and I was under the impression that you had to connect it through the IP address manually and that if there was an issue with that then you'd have to add the J1 profile as a non networked printer. But I guess I was just not reading carefully enough or miss interpreted! I'm going to share this with others in the group who are making this same mistake then. Thanks!

closed 2022-12-19 by alziepen

parachvte, 2022-12-19:
Actually, I don't have any Ultimaker printers, thus I didn't thought of users could add printer via the "Add a networked printer" feature.

Let me add some pics on the README to make it clearer...oh, maybe I'll do a little investigates to see if we could add networked printer directly.