Snapmaker/SnapmakerCuraPlugin #37: Cura 5.2.1 puts Snapmaker data into other printers' GCode

Miq1, 2022-12-31:
I have an Ender 3 printer configured in my Cura 5.2.1. As soon as I installed the J1 plugin (for my printer to come) Cura started putting J1 data into the generated GCode for the Ender - see below. F.i. the ``Printer`` and ``Extruder Mode`` lines are spoiled.
Most noticeable the preview plugin cannot be found any  more by the Ender's firmware.

```
;Header Start
;Version:1
;Slicer:CuraEngine
;Printer:Snapmaker J1
;Estimated Print Time:7230
;Lines:51458
;Extruder Mode:IDEX Full Control
;Extruder 0 Nozzle Size:0.4
;Extruder 0 Material:PETG
;Extruder 0 Print Temperature:240.0
;Bed Temperature:80
;Extruder(s) Used:1
;Work Range - Min X:98.475
;Work Range - Min Y:77.475
;Work Range - Min Z:0.2
;Work Range - Max X:136.525
;Work Range - Max Y:157.525
;Work Range - Max Z:5.96
;Thumbnail:data:image/png;base64,... cut out base64 data  ...
;Header End
;FLAVOR:Marlin
;TIME:7229
;Filament used: 2.95554m
;Layer height: 0.16
;MINX:98.475
;MINY:77.475
;MINZ:0.2
;MAXX:136.525
;MAXY:157.525
;MAXZ:5.96
;POSTPROCESSED
;Generated with Cura_SteamEngine 5.2.1
;
; thumbnail begin 230x180 6280
; /9j/4AAQSkZJRgABAQEAYABgAAD/2wBDAAgGBgcGBQgHBwcJCQgKDBQNDAsLDBkSEw8UHRofHh0aHB
```

parachvte, 2023-01-03:
You can choose to export G-code file, not "Snapmaker J1 G-code File".

[image]

closed 2023-01-03 by parachvte

Miq1, 2023-01-03:
Thanks, that was the solution.