Snapmaker/SnapmakerCuraPlugin #39: Filament runout detection for Octoprint?

evilC, 2023-01-13:
Is there any way to enable support for filament runout detection while printing via Octoprint?

parachvte, 2023-01-15:
Sorry I don't familiar with filament runout of OctoPrint. If someone can issue a PR or knowing how to, very appreciated.

evilC, 2023-01-16:
@parachvte this is not something I know much about, but in a [discussion on the FB group](https://www.facebook.com/groups/snapmakerj1/posts/886310342409279/?comment_id=886359992404314&__cft__[0]=AZXQqtqB-D6b-2n0Jb2KTf8NVqAQntnshYDBUvx3ivlRJyLFt9_oaPVrI0f9XJAs1C6xJNwF9kdFXLviPlgC3tvJ56LIQuVkzXa7ZtEs9OBOJpCqaW7_9OLyzjKsE6DgafgQahUJOolEizfeeFHoTAqKwwgD27PH_dkVRh2i3NduFS1Vsw5uksEgu5ktR2nZF70&__tn__=R]-R), someone seemed to know a bit about it

To quote:
> [Jeffrey Edgett](https://www.facebook.com/groups/326602605046725/user/637525433/?__cft__[0]=AZXQqtqB-D6b-2n0Jb2KTf8NVqAQntnshYDBUvx3ivlRJyLFt9_oaPVrI0f9XJAs1C6xJNwF9kdFXLviPlgC3tvJ56LIQuVkzXa7ZtEs9OBOJpCqaW7_9OLyzjKsE6DgafgQahUJOolEizfeeFHoTAqKwwgD27PH_dkVRh2i3NduFS1Vsw5uksEgu5ktR2nZF70&__tn__=R]-R)
The firmware supports the host action commands, but they are not yet enabled.
[Jade Wu](https://www.facebook.com/groups/326602605046725/user/100013471745505/?__cft__[0]=AZXQqtqB-D6b-2n0Jb2KTf8NVqAQntnshYDBUvx3ivlRJyLFt9_oaPVrI0f9XJAs1C6xJNwF9kdFXLviPlgC3tvJ56LIQuVkzXa7ZtEs9OBOJpCqaW7_9OLyzjKsE6DgafgQahUJOolEizfeeFHoTAqKwwgD27PH_dkVRh2i3NduFS1Vsw5uksEgu5ktR2nZF70&__tn__=R]-R), do you mind bringing this up with the devs? They just need to uncomment the `define HOST_ACTION_COMMANDS` line in Configuration_adv.h to allow the filament runout sensor, etc. to work with OctoPrint.

jledgett, 2023-01-17:
To be clear, this change would need to be made in the printers firmware — not the Cura plugin.  

Is there a repo yet for the firmware?

evilC, 2023-01-17:
> To be clear, this change would need to be made in the printers firmware — not the Cura plugin.
> 
> Is there a repo yet for the firmware?

No there is not. @parachvte has said it is OK to post these issues here for now.
I made a request for the source code, and was told that they will release it, but are "working to make it more stable for public use", which is technically a violationof the GPL license, but I'll cut them a bit of slack for now.