Snapmaker/SnapmakerCuraPlugin #48: Materials Have Wrong Retraction Settings

jledgett, 2023-02-24:
You have several retraction settings which are incorrect.   Look at any of the Polymaker materials.

The retraction distance is 6mm which is appropriate for a Bowden drive system, but not a direct drive unit like the J1.   Please update these to match Polymaker's suggested settings.

parachvte, 2023-02-24:
Please checkout our Snapmaker [material settings](https://github.com/Snapmaker/SnapmakerJ1CuraPlugin/tree/main/resources/materials), retraction distance is 1.5mm.

jledgett, 2023-02-24:
When will the others be updated (or removed for everyone's sanity 😆  )?

parachvte, 2023-02-27:
> When will the others be updated (or removed for everyone's sanity 😆 )?

We are not gonna change other materials, or another user is going to shout at us: Why retraction distance is so wrong for my CR-10 after installing your plugin? 😿 

I think by Cura's design, user should always manage their own materials, always select the right material and tune their parameters well.

parachvte, 2023-04-12:
> Is there something in the printer definition that indicates whether they will show up or not?

Actually they did fine tune material for specific machine: 

- [Example: Creality Printers use 5.0 mm retraction](https://github.com/Ultimaker/fdm_materials/blob/master/generic_pla_175.xml.fdm_material#L87)

This requires open pull request to Ultimaker's [FDM material repo](https://github.com/Ultimaker/fdm_materials), add machine specific settings.