Snapmaker/SnapmakerCuraPlugin #50: Artisan always prints using the left extruder

raleksandar, 2023-03-14:
I've just tested the v0.6.0 pre-release version with Artisan. I want to print a model using the right extruder, and even though I've configured that in Cura (it painted the model with the correct color in the preview), once the gcode is exported and loaded on Artisan, the model was printed using the left extruder.

I have tried disabling the left extruder in Cura, but that made no difference. What else can I try?

sumibin, 2023-03-17:
Make sure you are not using a left extruder in your extruder setup. Otherwise, this is what happens in the picture above.

[image]

raleksandar, 2023-03-17:
Checked that before posting.
Also, when I disable the left extruder Cura hides all options where you can configure the extruder to use (the pictured option is not available when only one extruder is enabled in `Settings > Left/Right Extruder > Disable Extruder`)

tensile23, 2023-03-30:
I think it's a cura usage issue. When you disable an extruder, the resulting Gcode will have only one T1 instruction.
In addition, you are required to upgrade the firmware of Artsian to the latest. （(In older firmware, G28 causes the activation nozzle to switch to T0)
[image]

[image]

closed 2023-10-16 by tensile23