Snapmaker/SnapmakerCuraPlugin #62: Printing an object with two hotends: initial temperatures mixed between hotends

Braintoe, 2023-03-25:
When slicing the attached object for the Snapmkaer J1 which uses
- left hotend with PETG to print the part (start temperature 240°C)
- right hotend with PLA to print the supports (start temperature 230°C, supposed to drop to 215°C after the first layer))
- bed temperature set to 75°C which is defined by the left hotend (this works correctly)

the code created with the Cura plugin correctly shows the correct values in the header:

`;Extruder 0 Nozzle Size:0.4
;Extruder 0 Material:PETG J1 anthrazit
;Extruder 0 Print Temperature:240
;Extruder 1 Nozzle Size:0.4
;Extruder 1 Material:PLA J1 blaumetallic
;Extruder 1 Print Temperature:215`

but the printer ends up mixing temperatures between hotends due to the gcode being wrong. It heats the right hotend (no.1) first to 240°C:

`T1
M82 ;absolute extrusion mode
;--- Start G-code Begin ---
M104 S240 ;Set Hotend Temperature
M140 S75.0 ;Set Bed Temperature`

and a little later then heats up the left hotend to 215°C:

`;LAYER:0
M107 P1
M104 T0 S215`

Please find tghe files attached - the Zip file contains the gcode and a 3MF Cura project that contains all the defined settings. If I did not overlook anything, the issue should be reproducible by simply opening the 3MF file as a project and slicing it in Cura.

[Distanzstueck.zip](https://github.com/Snapmaker/SnapmakerCuraPlugin/files/11069118/Distanzstueck.zip)

Tested with Cura 5.3.0.

parachvte, 2023-05-02:
@Braintoe 

Sorry for the late response. It took some time for me to understand how this could happen.

The reason that your project uses left extruder (T0) initial temperature as follows:

- Cura uses function `ExtruderManager.getInitialExtruderNr()` ([code here](https://github.com/Ultimaker/Cura/blob/main/cura/Settings/ExtruderManager.py#L306)) to determine which extruder is the initial extruder. Priorities as: 1) adhesion extruder 2) support extruder 3) first used extruder to print mesh
- In your case, the G-code starts to print with right extruder (T1), which is the skirt of support. But in the settings the adhesion extruder is "Not overridden", in the `getInitialExtruderNr()` function Cura thinks it's the left extruder.

I think this is an issue of inconsistency of Cura and CuraEngine, Cura uses a different strategy to determine the initial extruder other than CuraEngine.

**To solve your problem**, explicitly specify the adhesion extruder to left extruder (240°C) or to right extruder (230°C) can make Cura choose the correct temperature for initial layer.

[image]

Braintoe, 2023-05-05:
Thanks a lot for your detective work! Let's see if /when the Cura team solves that. I have other issues with Cura as well.

I solved the problem in the meantime by using Prusaslicer instead - that one works flawlessly :-)

closed 2023-05-12 by parachvte