Snapmaker/SnapmakerCuraPlugin #72: J1 plugin for Cura

oSoDirty, 2023-04-29:
No matter what printer is selected it prints defaults from the J1 to the gcode when exported. Not sure if it's a real issue, but I imagine that if it thinks it's the wrong printer, it could send some wrong settings.

parachvte, 2023-05-02:
Sorry I can't understand your problem. What do you mean by "it prints defaults from the J1 to the gcode when exported"?

If you are using old J1 plugin and Snapmaker Plugin together, it could lead to problems. As we stated:

```
If you have already installed the Snapmaker J1 Plugin, make sure that you uninstall it before installing or updating Snapmaker Plugin.
```

Please uninstall the old J1 plugin, and then install the new one and re-add printers (J1).

oSoDirty, 2023-05-03:
I'm just saying that no matter what printer I have selected in Cura, the
gcode comments in the export that the printer in use is the J1. I only use
J1 addon, I just added it in preparation since I went ahead and ordered a
J1. I will remove and try t Snapmaker addon instead.

Example:

;Header Start
;Version:1
;Slicer:CuraEngine
;Printer:Snapmaker J1
;Estimated Print Time:10703
;Lines:334698
;Extruder Mode:IDEX Full Control
;Extruder 0 Nozzle Size:0.4
;Extruder 0 Material:SS TPU
;Extruder 0 Print Temperature:218.0
;Bed Temperature:81.0
;Extruder(s) Used:1

This is printed in gcode when I slice for my Tevo or Creality.

Probably a completely harmless bug, just thought I'd point it out. Can't
wait to try the J1.

oSoDirty, 2023-05-03:
New app doesn't have this effect. Thanks for the tip.

parachvte, 2023-05-04:
There are several output format options:

[image]

In our previous plugin (SnapmakerJ1Plugin), we mis-used "Snapmaker J1 G-code File" over the default option "G-code File", you will need to select "G-code File" manually when you want to export as default format.

closed 2023-05-04 by parachvte