Snapmaker/SnapmakerCuraPlugin #75: doesn't send files over wifi and does not generate thumbnail preview

colombod, 2023-05-03:
With latest version of cura (5.3.1) the plugin is not giving me the option to send file via network and there is no thumbnail preview.

parachvte, 2023-05-12:
- Which printer are you using?
   - You need to add a non-networked printer that pre-defined after installing the plugin. 
- Select "Snapmaker Flavor G-code File (*.gcode)" when exporting G-code

colombod, 2023-05-12:
A350 profile installed and slicing. Doesn't show any other option other than saving to disk

colombod, 2023-05-12:
How do I send this gcode to the machine?

parachvte, 2023-05-12:
Could you try out [version 0.9.2](https://github.com/Snapmaker/SnapmakerCuraPlugin/releases/tag/0.9.2)?

In previous version (0.9.0), only Snapmaker 2.0 printers with dual extruder can be found via network.

colombod, 2023-05-12:
how do I install the pre-release package?

parachvte, 2023-05-12:
[image]

There is a section in README, drag downloaded [SnapmakerPlugin-0.9.2.curapackage](https://github.com/Snapmaker/SnapmakerCuraPlugin/releases/download/0.9.2/SnapmakerPlugin-0.9.2.curapackage) into Cura Window and restart Cura will work.

colombod, 2023-05-12:
should have read first , thank you :)

closed 2023-05-12 by colombod