Snapmaker/SnapmakerCuraPlugin #77: Cant connect to printer through wifi

pwvandeursen, 2023-05-30:
I have installed the cura plugin, added the snapmaker 2 printer and then restarted. added a project, sliced, but no option to send to the printer directly.

Do I need to find the printer in my network first somewhere?

pahenning, 2023-06-11:
Same questions from my side: How does one teach Cura the IP address of the SnapMaker? If I try to find it on the network during printer installation, Cura is unable to connect.

NivO5, 2023-07-06:
I have it too, tested from 2 different PCs, and checked windows Firewall settings. The 3rd party plugins work like a charm.

NivO5, 2023-07-10:
Had it too, upgrading the plugin to 0.9.2 solved it for me.

closed 2023-10-16 by tensile23

chinloon, 2025-02-13:
I am unable to connect to my snapmaker 2 A350 over wifi.  DIscovery doe snot work and there is no connections setting to add IP address when printer is added as local.