Snapmaker/SnapmakerCuraPlugin #83: Does not wait for temp on extruder change.

hilmerhall, 2023-10-15:
Currently when switching nozzles for dual extruder (Artisan) it starts printing before new nozzle gets to temp.

tensile23, 2023-10-16:
Can you please provide us with the Cura project and the Gcode so we can confirm the problem?

hilmerhall, 2023-10-16:
Sure. I noticed it by watching the machine's screen. It started to print way before nozzle got to temp on switch.
I added M109 T0 S{material_print_temperature} in the extruder start code in machine settings which does the trick. (i will try adding a z up down before after).
otherwise here are some files.
[requested_files.zip](https://github.com/Snapmaker/SnapmakerCuraPlugin/files/12915086/requested_files.zip)

tigercjn, 2024-04-10:
The Extruder data is missing the start Gcodes under each extruder setting.